cetic/6lbr #414: webserver buffer overflow when processing requested url

sploitem, 2021-05-27:
if u enter url like this:
[bbbb::100]/AAAAAAAAAAAAAAAAA.....AAAAAAAA (len = 1000)

it will crash with stack smashing.

because of strcat - https://github.com/cetic/6lbr/blob/develop/examples/6lbr/apps/6lbr-webserver/httpd.c#L119

sploitem, 2021-05-28:
i think this is exploitable on arm debian since there is no stack smashing detection.

SalvaHasan, 2021-09-20:
@sploitem I'm facing the same issue, 6lbr.log : `HTTP: File '/favicon.ico' not found` and  `WEB: Buffer overflow`
which results in 6lbr crashing every few minutes. Have you found any solutions?

sploitem, 2021-09-20:
> @sploitem I'm facing the same issue, 6lbr.log : `HTTP: File '/favicon.ico' not found` and `WEB: Buffer overflow`
> which results in 6lbr crashing every few minutes. Have you found any solutions?

On my machine `HTTP: File '/favicon.ico' not found` and `WEB: Buffer overflow` didn't result in crash.

SalvaHasan, 2021-09-20:
In my case the [bbbb::100] becomes unreachable as soon as buffer overflows appears in 6lbr.log

sploitem, 2021-09-21:
> In my case the [bbbb::100] becomes unreachable as soon as buffer overflows appears in 6lbr.log

I don't think this issue is a root cause of your crash, 'cause this issue requires a request with a long url like this: [bbbb::101]/AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA.html

Check the 6lbr log for operation due to it's restart.

sploitem, 2021-09-21:
Also i suggest you to use contiki-ng instead this one.

SalvaHasan, 2021-09-22:
> I don't think this issue is a root cause of your crash, 'cause this issue requires a request with a long url like this: [...]

Yeah I don't enter any long URL. Mine crashes when I start to navigate the tabs (sensors, config, admin etc.) on the webpage and the log only shows Web: buffer overflow

How can I use 6lbr from contiki-NG? I configured cetic 6lbr this using [this guide](https://sunmaysky.blogspot.com/2018/04/build-and-run-6lbrcontiki-ng-cc26xx-web.html?m=1) and built it using `make WITH_CONTIKI=0 CONTIKI=~/contiki-ng WERROR=0 all`.
Is there any proper guide to use 6lbr from contiki-NG?

sploitem, 2021-09-22:
Your 6lbr was built with contiki or contiki-ng ?
Which page\tab causes a crash ?

SalvaHasan, 2021-09-22:
> Your 6lbr was built with contiki or contiki-ng ?
> Which page\tab causes a crash ?

It was built using contiki-ng and the crash was random it wasn't caused by any specific tab.
I changed the 6lbr interface config from bridge mode to routing mode, 6lbr.conf: `RAW_ETH=0
BRIDGE=0
DEV_ETH=eth0
DEV_TAP=tap0`
This fixed the issue for me.

sploitem, 2023-12-31:
CVE-2021-46901